ros2/rcl_interfaces #2: IDL parsing error

wjwwood, 2015-05-01:
When using the `review_results` branch on the farm (on Windows) I get an error with IDL parsing:

http://54.183.26.131:8080/job/ros2_batch_ci_windows/216/console
```
(CustomBuild target) -> 
         CUSTOMBUILD : IDL parser error : opening file C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\rcl_interfaces\rosidl_generator_dds_idl\rcl_interfaces\dds_opesplice\SetParametersResponse_.idl
```

I haven't tried the Linux or OS X batch jobs yet.

wjwwood, 2015-05-05:
This is not a problem on Linux or OS X.

esteve, 2015-05-05:
Have you looked into why Windows uses `opesplice` instead of `opensplice`?

wjwwood, 2015-05-05:
I can't find any reference to that misspelling in the code base. I suppose it could be happening _in_ the OpenSplice code generator, but that's bizarre. I'm a bit at a loss.

tfoote, 2015-05-05:
New with parameters. Can't isolate the change which has caused this regression.

tfoote, 2015-05-05:
Brainstormed idea: There might be an issue with the regex splitting the srv files?

wjwwood, 2015-05-06:
The weirdness continues, this time the `_` is missing:

http://54.183.26.131:8080/job/ros2_batch_ci_windows/229/console
```
"C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\rcl_interfaces\rcl_interfaces__rosidl_typesupport_opensplice_cpp.vcxproj" (default target) (14) ->
       (CustomBuild target) -> 
         CUSTOMBUILD : IDL parser error : opening file C:\Jenkins\workspace\ros2_batch_ci_windows\workspace\build\rcl_interfaces\rosidl_generator_dds_idl\rcl_interfaces\ddsopensplice\SetParametersResponse_.idl
```

Maybe the missing character is a red herring, but it certainly is suspicious.

wjwwood, 2015-05-06:
Also, that last test was without @dirk-thomas's regex change, so that's not the issue, I think.

wjwwood, 2015-06-02:
This is still present after reinstalling Windows. It does not occur when building in the cmd prompt, but does with Jenkins, so there's something going on there still. Also, it seems to happen on the Services but I'm not sure if that's indicative of anything. The build still fails eventually, but that's due to another syntax error issue in the new `test_communications` package.

There are still (seemingly) random characters missing from the file path it is trying to open... At this point I'm open to ideas.

closed 2015-06-15 by wjwwood