ros2/rcl_interfaces #7: bool as a return code

BobDean, 2015-09-19:
I have been debating bringing this up for a while now, hopefully I have learned enough from the parameter discussion interactions to properly describe my general design concern. My opinion, based on the last 20 years of professional experience, is that bool should never be used as a return code (i.e. SetParametersResult.msg), An int ErrorCode should be used instead, similar to how the posix c interfaces always return an error code (or set errno, but thats a separate discussion). 

There are a couple reasons for this:
1) an error code may encode more semantic information into the value than a bool can. For example with setting parameters there are specific success/failure cases which can be identified: value set successfully, comms failure, comms timeout, parameter validation failed, authentication failed (you do not have permission to set this value), (there are probably a few more).  

This allows the client software to perform meta-cognitive analysis on the result and automate responses or interact with the user in more intelligent manner. Additionally, extraction and analysis of data logs become more powerful within matlab and other tools as the default semantic content can actually be graphed over time (it's pretty cool).  how can a string be graphed?

2) if at some time in the future the change from bool to error code needs to occur, it is possible to introduce very hard to find bugs in your logic because if(msg.success) may evaluate to true when it should be false, leading to hours of head scratching wondering why code segment X no longer runs. 

3) the use of a string reason is good for additional information, however relying on it by itself puts us at the whims of the implementer. 

4) support for string locale descriptions of the error.

5) consistency with api calls or messages which do have a non bool return/success value.

jack-oquin, 2015-10-15:
+1 to this proposal

clalancette, 2017-08-04:
I can see where you are coming from with this.  An integer return result concerns me a bit in that the mapping of semantic name -> integer could be a bit tricky, but I think we can do the same thing we do sensor_msgs/msg/Range.msg, where we define the enum in the message itself.  @ros2/team What do you think about changing the return type here to an int?

wjwwood, 2017-08-07:
Sorry we let this languish for so long, thanks @clalancette for reviving it. I'll try to address it inline incase there is still some interest in this topic, but in general I'm -0 for changing it to an int. Without a suggested set of error codes to predefine I think the error code adds very little benefit or even perhaps adds confusion. With a reasonable set of error codes suggested I would be in favor of a change like this (though in addition to the reason string, not in place of it).

When we were making this interface originally I believe we left it bool because we couldn't come up with a reasonable (or maybe "useful") set of error codes that would be common and expressive enough to include.

> an error code may encode more semantic information into the value than a bool can. For example with setting parameters there are specific success/failure cases which can be identified: value set [...]

While that's true, many of the other possible examples you gave are related to communications and not setting the parameter. If you have a comms problem or you are restricted access to the comms then the middleware should notify you of that out-of-band.

So the return code would need to encapsulate something specific to why the parameter was not set. It could be that we could come up with a few common issues, but that's in part why the reason string is there, since that information was intended to be set by the parameter server, who could include as much or little information as they needed. You only really need to have a error code if you expect to programmatically react to the failure some how. The fact that POSIX uses them has more to do with efficiency, though there are cases where you can recover on specific error codes.

> Additionally, extraction and analysis of data logs become more powerful within matlab and other tools as the default semantic content can actually be graphed over time (it's pretty cool). how can a string be graphed?

Again that's only useful if you propose a predefined set of return values. If I define `1` as "permission denied" and you define it as "out of range" then it's not any more useful than the string. The string can be graphed by using a set to collect the messages and graphing the index (perhaps in a few cases matching just part of the string). The string's aren't useless and I would argue they are more useful more often for setting parameters because they can include more information, e.g. a return code which means "invalid value" versus "camera exposure value '0.1' is invalid because it is longer than the requested publish rate's period of '0.05'".

> the use of a string reason is good for additional information, however relying on it by itself puts us at the whims of the implementer.

I agree, but the same is true of the error codes (imo) unless we have a predefined set of error codes.

> consistency with api calls or messages which do have a non bool return/success value.

I don't feel these messages are a departure from any existing stuff, perhaps you could give some examples.

> extensibility by other developers. perhaps I have a requirement for a customer which could be solved by adding an extra error code.

Within a single organization this might be fine, but we're striving for these messages to be interpreted universally, so the reason string, while less machine readable, is unambiguously interpretable (simply print the string to the screen or log). Imagine a general tool that renders these responses in a GUI, how would it interpret the custom return codes a company defined? Just print the number? That doesn't give much information to the user. Again, I'd be ok with this if there were also some common error codes (invalid value, invalid range, permission denied, etc...) which the tool could interpret and give the user more info on than just a number, but only to augment the reason string.

BobDean, 2017-08-07:
sorry, i will not be able to respond further.

clalancette, 2017-08-07:
@BobDean Thanks for the response.  Given @wjwwood 's response, and given that you aren't too interested in this right now, I'm going to close this issue out.  If you have time to discuss further sometime in the future, feel free to reopen this.

closed 2017-08-07 by clalancette