ros2/rcl_interfaces #19: Remove double-specification of default values in ParameterValue msg

dhood, 2017-08-22:
Default values were added explicitly in https://github.com/ros2/rcl_interfaces/pull/18 to prevent a valgrind error of uninitialised fields during serialisation. When messages can be created as pre-initialised with default values in client libraries, revert that PR and use default-initialised messages instead (to prevent specification of the default values in multiple locations).

mikaelarguedas, 2017-09-28:
#18 Needs to be reverted once https://github.com/ros2/ros2/issues/396 related PRs are merged.
@clalancette FYI, not sure if you want to "connect" them together

clalancette, 2017-10-02:
@mikaelarguedas Thanks for the heads up; I'll connect this up with #396 as you suggest.

PR #24 (2017-11-01): Revert "Provide defaults for ParameterValue fields (#18)"
This reverts commit d25c83b657db452c6abf51aacf5a6793c3941473.

Fixes #19 
connects to ros2/ros2#396
@clalancette FYI

Same valgrind test as in https://github.com/ros2/rclcpp/pull/358
```
==4814== HEAP SUMMARY:
==4814==     in use at exit: 96,348 bytes in 48 blocks
==4814==   total heap usage: 5,768 allocs, 5,720 frees, 7,413,891 bytes allocated
==4814== 
==4814== Searching for pointers to 48 not-freed blocks
==4814== Checked 634,008 bytes
==4814== 
==4814== LEAK SUMMARY:
==4814==    definitely lost: 0 bytes in 0 blocks
==4814==    indir [... cut]
closed 2017-11-01 by mikaelarguedas