ros2/rcl_interfaces #31: Include path order references underlay before overlay

ayrton04, 2018-01-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - Binaries with an overlay containing `rcl_interfaces`
- Version or commit hash:
  - `param-arrays` branch on github.com:ayrton04/rcl_interfaces.git

#### Steps to reproduce issue
```
mkdir -p ~/ament_test/src
cd ~/ament_test/src
git clone git@github.com:ayrton04/rcl_interfaces.git -b param-arrays
cd ..
source /opt/ros/ardent/setup.bash
ament build
```

#### Expected behavior
Successful build

#### Actual behavior
```
[ 83%] Building C object CMakeFiles/rcl_interfaces__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c.o
/home/automatom/ros2/build/rcl_interfaces/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c: In function ‘rcl_interfaces_parameter_value__convert_from_py’:
/home/automatom/ros2/build/rcl_interfaces/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c:111:61: error: ‘rcl_interfaces__msg__ParameterValue {aka struct rcl_interfaces__msg__ParameterValue}’ has no member named ‘byte_array_value’
 if (!rosidl_generator_c__byte__Array__init(&(ros_message->byte_array_value), size))
```

#### Additional information
Please see [this](https://answers.ros.org/question/278910/ros2-ament-overlay-issue-with-messages/) ROS answers question for more details.

FYI @wjwwood

wjwwood, 2018-01-09:
From my answer...

This is what the cc line looks like on my machine:

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-Drcl_interfaces__rosidl_typesupport_introspection_c__pyext_EXPORTS
-I/Users/william/ament_test/build/rcl_interfaces/rosidl_typesupport_c
-I/Users/william/ament_test/build/rcl_interfaces/rosidl_typesupport_introspection_c
-I/Users/william/ros2_ws/install/include  <-- this has the old header
-I/Users/william/ament_test/build/rcl_interfaces/rosidl_generator_c  <-- this has the new header
-I/Users/william/ament_test/build/rcl_interfaces/rosidl_generator_py
-I/usr/local/Cellar/python3/3.6.3/Frameworks/Python.framework/Versions/3.6/include/python3.6m

-fPIC

-Wall
-Wextra
-o
CMakeFiles/rcl_interfaces__rosidl_typesupport_introspection_c__pyext.dir/rosidl_generator_py/rcl_interfaces/srv/_describe_parameters__response_s.c.o

-c
/Users/william/ament_test/build/rcl_interfaces/rosidl_generator_py/rcl_interfaces/srv/_describe_parameters__response_s.c
```

The problem is likely in the generator code and the order in which it adds headers to the include path with cmake.

mikaelarguedas, 2018-01-27:
@ayrton04 see https://github.com/ros2/rosidl/pull/261 for a fix.

cc line before the patch:
```
/usr/bin/cc  -Drcl_interfaces__rosidl_typesupport_c__pyext_EXPORTS 
  -I/root/ament_test/build/rcl_interfaces/rosidl_typesupport_c
  -I/opt/ros/ardent/include
  -I/root/ament_test/build/rcl_interfaces/rosidl_generator_c
  -I/root/ament_test/build/rcl_interfaces/rosidl_generator_py
  -I/usr/include/python3.5m

  -fPIC   -Wall -Wextra -o CMakeFiles/rcl_interfaces__rosidl_typesupport_c__pyext.dir/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c.o   -c /root/ament_test/build/rcl_interfaces/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c
```

cc line after the patch:
```
/usr/bin/cc  -Drcl_interfaces__rosidl_typesupport_c__pyext_EXPORTS
  -I/root/ament_test/build/rcl_interfaces/rosidl_generator_c
  -I/root/ament_test/build/rcl_interfaces/rosidl_generator_py
  -I/usr/include/python3.5m -I/root/ament_test/build/rcl_interfaces/rosidl_typesupport_c
  -I/opt/ros/ardent/include

  -fPIC   -Wall -Wextra -o CMakeFiles/rcl_interfaces__rosidl_typesupport_c__pyext.dir/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c.o   -c /root/ament_test/build/rcl_interfaces/rosidl_generator_py/rcl_interfaces/msg/_parameter_value_s.c
```

ayrton04, 2018-01-29:
Awesome, thanks!

mikaelarguedas, 2018-01-29:
fixed by ros2/rosidl#261

closed 2018-01-29 by mikaelarguedas