ros2/rcl_interfaces #50: rosgraph_msgs Log message type.

mlautman, 2018-11-14:
I am working on a port of [rqt_console](https://github.com/ros-visualization/rqt_console) with @dirk-thomas which uses the `rosgraph_msgs/Log` message. I see that `rosgraph_msg/Log` hasn't been ported to ros2 yet and I was wondering if I should implement the Log msg or if there was a specific reason for not porting Log in rosgraph_msgs. As this is more of a design decision, I am happy to move the conversation to discourse if you believe that this should be a more general discussion.

tfoote, 2018-11-14:
I don't believe that there's any reason that it hasn't been ported except that it hasn't been needed yet. 

I would suggest coordinating with @nburek

He's working on several PRs right now to start publishing console outputs: 
https://github.com/ros2/rclcpp/pull/582
https://github.com/ros2/rcl/pull/327
https://github.com/ros2/rcutils/pull/127

nburek, 2018-11-14:
Hey Mike,

I am currently working on adding support for logging to a "rosout" topic and as part of that have ported over the Log message from rosgraph_msg. The message I'm working with looks mostly the same except that I've adjusted the level constants to match the enum values from the rcl code and I've removed the list of topics in the message as it didn't make sense to me that you would want that list with every log message. 

I was also looking at moving the message from "rosgraph_msg" into "common_interfaces/diagnostic_msgs" as logs seemed better grouped as a diagnostics tool than a rosgraph tool. 

I'll start working on getting a pull request created for my Log message port (it may take a couple of days to go through my company's open source approval process before I can officially make the pull request).

dirk-thomas, 2018-11-14:
Since the message should be used by `rcl` please add the Log message to [rcl_interfaces](https://github.com/ros2/rcl_interfaces/tree/master/rcl_interfaces) package. That one is a dependency of `rcl` already. Putting the message in `common_interfaces` would add another dependency to `rcl` and require `common_interfaces` to be released before `rcl` which isn't desired.

nburek, 2018-11-14:
Ok, makes sense. I'll add it to rcl_interfaces instead.

dirk-thomas, 2018-11-14:
> For consistency I'd suggest keeping the same package + name too.

Imo `rosgraph_msgs` doesn't make any sense for the `Log` message. So I don't think keeping that package name is a good idea.

brawner, 2020-02-06:
@mlautman @nburek  This issue can be closed with #53 merged, yah?

closed 2020-02-06 by dirk-thomas