ros2/rcl_interfaces #58: extract message / service / action files from test_msgs package

dirk-thomas, 2019-01-14:
Currently many packages (mostly message generators) have their own test interface to check if the generator works as expected. But each package has an arbitrary subset. And with the amount of corner cases and variations it would be helpful if all these interface files exist in a single location. The message generator packages can't depend on `test_msgs` since that package itself depends on all message generators.

To avoid the duplication and get rid of the differences of these interface files I would like to propose to extract the actual interface files from the `test_msgs` package into a new separate package (and repository). I would propose the package / repo name `test_interface_files`. The `test_msgs` package would then be updated to still invoke the message generation for all these files - just not contain the interface files anymore but get them from the new package.

I am mostly looking for feedback on this before actually spending the time to create PRs for it.

jacobperron, 2019-01-15:
+1 for having a single place for test interface files.

It's unfortunate that we can't keep them alongside `test_msgs` though, since they are closely coupled. Maybe the two packages could be in the same repo for convenience? E.g. Move `test_msgs` out of this repo into another with `test_interface_files`.

clalancette, 2019-01-15:
Note that the individual `rosidl` generators *also* have their own `.msg` files to ensure generation happens properly (see https://github.com/ros2/rosidl/tree/master/rosidl_generator_cpp/msg and https://github.com/ros2/rosidl_python/tree/master/rosidl_generator_py/msg).  It may make sense to consolidate these into `test_interface_files` as well.

dirk-thomas, 2019-01-15:
> Note that the individual `rosidl` generators also have their own `.msg` files to ensure generation happens properly

There are the duplicates I was referring to.

dirk-thomas, 2019-02-09:
Not working on this until after the IDL changes have landed.

jacobperron, 2019-04-19:
One issue with this approach is that it won't be able to capture tests involving nested types from different packages (e.g. [NestedMessage.action](https://github.com/ros2/rcl_interfaces/blob/1189517a0217a2cb9f78e71c2959542429ad9fc0/test_msgs/action/NestedMessage.action#L4)).

jacobperron, 2019-04-19:
@dirk-thomas I've built on top of your changes and opened a PR for the new package `test_interface_files` https://github.com/ros2/test_interface_files/pull/1.

jacobperron, 2019-04-24:
I've started going through our packages that have test interfaces and switching them over to use the new `test_interface_files` package:

- [ ] rosidl_generator_c
- [ ] rosidl_generator_cpp ros2/rosidl#367 (ready for review)
- [ ] rosidl_runtime_py
- [ ] test_msgs

dirk-thomas, 2019-04-25:
This work will conflict with the ongoing work on WString support (ros2/rosidl#352). I just wanted to mentioned it early that we don't run into problems last minute before the API freeze date and try to land both at the same time.

jacobperron, 2019-04-25:
Thanks for the heads up. This ticket is lower priority and can be done piecemeal (ie, we can update affected packages independently), so I can give the WString support right of way.

jacobperron, 2019-04-25:
Another potential conflict: https://github.com/ros2/rcl_interfaces/pull/68

closed 2019-10-04 by jacobperron