ros2/rcl_interfaces #71: add lower / upper boundary to parameter descriptor

dirk-thomas, 2019-04-17:
Dynamic reconfigure in ROS 1 allows to define a lower / upper bound for numeric parameters. That information enables more usable UI elements like sliders.

While the logic to check if a specific parameter can be arbitrary complex narrowing the valid numeric values with a lower / upper bound already covers many use cases. Therefore the [ParameterDescriptor](https://github.com/ros2/rcl_interfaces/blob/1189517a0217a2cb9f78e71c2959542429ad9fc0/rcl_interfaces/msg/ParameterDescriptor.msg) should be extended to add this information.

For integer parameters obviously integer boundaries would be fitting. For floating point number the boundaries should be floating point numbers though in order to e.g. express a range [0, pi].

Since any fields are part of the message independent of the actual parameter type adding multiple different boundaries for each type isn't appealing since it increases the message size.

Therefore I propose to add lower and upper bounds with a floating point type. It enables the precise range necessary for floating point parameters and for integer parameters the conventions could be to round to the closest integer.

I am looking for early feedback on this so that the implementation can start beginning of next week.

wjwwood, 2019-04-18:
Obviously we don't need to sort out or act on everything in this comment below this line, but I just wanted to brain dump so we might consider these things while working on this and perhaps avoid needing to make more painful API breaks in the future.

----

I think it would be worth having a step as well, so you could do things like `[0.0, 1.0, 0.1]` or `[0, 100, 2]` for even numbers only. Dynamic reconfigure supported this.

The step could be documented that it needs to be unsigned and a value of `0` could be used to disable it for floating point and integers (though for integers 1 would be the same as 0).

A step value complicates verification a bit (step needs to be `<= distance(min, max)`), but you already need to check for special cases like: `min == max`, or if you want to support things like `[1, -1, ...]` (inverted slider) or `[1, 1, ...]` (only one valid value), so checking that the step is a valid value is not much more of a burden. It does, however, lead to interesting test cases like `[0, 1, 2]`.

----

> Since any fields are part of the message independent of the actual parameter type adding multiple different boundaries for each type isn't appealing since it increases the message size.

I agree having one for int and float is wasteful, but I'm not sure how important that is because I believe  this descriptor is going to be sent infrequently (it is not part of the parameter event payload for example). And so I think we should consider simplicity and expressiveness as part of the trade-off too.

----

In the interest of making progress on this asap, I'd like to propose this concrete addition to the existing message, in order to have something to pick apart:

```
uint8 CONSTRAINT_INT_RANGE=0
uint8 CONSTRAINT_FLOAT_RANGE=1
# ....

# Sequence of constraints which apply to the parameter.
uint8[] constraints

# Constrains possible integer values to [lower, upper, step], inclusively.
# Lower may be <=> upper.
# Step is considered unsigned, and will be treated as `abs(step)`.
# A step value of `0` implies no discrete step.
# The step value must be <= the distance between upper and lower.
int64[3] integer_range_constraint = [0, 0, 0]

# Constrains possible floating point values to [lower, upper, step], inclusively.
# Lower may be <=> upper.
# Step is considered unsigned, and will be treated as `abs(step)`.
# A step value of `0` implies no discrete step.
# The step value must be <= the distance between upper and lower.
float64[3] float_range_constraint = [0.0, 0.0, 0.0]
```

Points for discussion off the top of my head:

- Is `CONSTRAINT`/`constraint` the right word/name for this?
  - For me this is to separate them from other things which are "constraint like", e.g. `read_only`, or things that are definitely not constraints, but could be considered meta data, e.g. `description`.

- Is there any situation where more than one constraint would apply?
  - Obviously not these two examples (int and float range), but in the future.
  - Might be needed if you consider `read_only` to be a constraint, then you'd need to express `read_only` and a range? Or maybe `read_only` is always expressed (never type conditional).
  - I had imagined a `string[<=256] additional_constraints` which could be used by the programmer to convey complex constraints in plain English, so they could be displayed in the GUI at least, e.g. `only prime numbers`. If consider this message (or note) as a constraint as well, then it would need to be listed at the same time, or we could instead have it just be a field and use empty string as "not set".
  - The upside to a sequence of `uint8` for `constraint(s)` over a single value is that it's future proof and only cost a little (a sequence of size 1 is not significantly larger than a single uint8).
  - The downside is that we now need to validate it for edge cases like: repeated entries, mutually exclusive entries (no int and float range at the same time), etc...

- Should there be a separate int and float range?
  - I think it's not that expensive given the context in which the message is used, but I could be wrong about that.
  - I think it's more explicit, simpler to document, and simpler to use on both sides in code.

- Should there be a step, or just upper/lower bounds?

- Should each "constraint" be a separate message type (include int and/or float range)?
  - These ranges can live as `vec[3]`, but would benefit from step being unsigned in the integer case.
  - Other types of constraints may require multiple fields to express and therefore may benefit from encapsulation, e.g. "int64 choice" might need an associative array with `string[] choices` and `int64[] values`. We obviously don't have to sort the specifics out now, but if we think it may be needed, it might be neater to organize each constraint into its own msg definition.

----

For reference:

- Dynamic reconfigure tutorials: https://wiki.ros.org/dynamic_reconfigure/Tutorials
  - https://wiki.ros.org/dynamic_reconfigure/Tutorials/HowToWriteYourFirstCfgFile
  - https://wiki.ros.org/dynamic_reconfigure/Tutorials/SettingUpDynamicReconfigureForANode
- Dynamic reconfigure msg's for communicating between the server (node) and client (gui):
  - https://docs.ros.org/melodic/api/dynamic_reconfigure/html/index-msg.html
  - https://github.com/ros/dynamic_reconfigure/tree/melodic-devel/msg
- https://github.com/pal-robotics/ddynamic_reconfigure_python (yes dynamic dynamic reconfigure)

tfoote, 2019-04-18:
I think that considering them to be constraints makes sense. And I think that having multiple constraints could relatively easily be possible if we start supporting more directions. I would suggest that we consider having each constraint be it's own class and the Constraints message just has a dynamic list of each type. For evaluation you simply iterate the each list. An empty list is immediately passed. 

Strawman ParameterConstraints.msg
```
ReadOnly read_only_constraint []
FloatingPointRange floating_point_range_constraint [] 
IntegerRange integer_range_constraint []
...
```
A simple one like read only would be: 

ReadOnly.msg
```
bool read_only
```

A more complex one like integer range:
IntegerRange.msg
```
int min
int max
int step # if greater than 0 step by this amount, otherwise assume step size 1.
```
And the different types can have different logic for multiple elements [1,2] [4,5] could be defined to evaluate as supporting the union of the constraints since it doesn't make sense to do the intersection of ranges that could just be defined as a range. Other constraint types could support specific logic for combining. 

FloatingPointRange.msg
```
float64 min
float64 max
```

And going from there we can add new datatypes to the Constraints message and when they are added it needs to propgate to all the implementations. 

Other potential choices are StringChoice and IntegerChoice, maybe StringLength.

sloretz, 2019-04-18:
We might be able to take some inspiration from [`<restriction>` in XML Schema](https://www.w3schools.com/xml/schema_facets.asp). The restrictions that apply to numeric types are:

* enumeration
* fractionDigits
* maxExclusive
* maxInclusive
* minExclusive
* minInclusive

I agree with @wjwwood's idea of having a sequence of constraints with an enum for constraint types, and @tfoote's idea of the constraints being classes instead of arrays. Together those allow adding the bare minimum now with more constraints in the future.

Instead of having one class with two fields `min` and `max`, I think we can borrow the idea from XMLSchema that `min` and `max` are independent constraints. Assuming there needs to be validation code for each constraint class, I think separating them could simplify the validation code. The `min` constraint can be checked without caring if there is also a `max` constraint.

Even `step` can be its own constraint. It just needs one valid value; it doesn't necessarily have to be the min or max.

```
float64 step
float64 start
```

dirk-thomas, 2019-04-18:
> I think it would be worth having a step as well...

:+1: Absolutely.

> you could do things like [0.0, 1.0, 0.1]

Storing these values without any semantic value in an indexed array seems to be difficult to read. I think storing them in explicitly named fields would be better.

> I would suggest that we consider having each constraint be it's own class and the Constraints message just has a dynamic list of each type. 

I like the idea of using lists since it nicely solves the problem knowing which constraints are "active". For now limitting the number of constraints of the same type can easily be limited to 1. I don't think we need to support multiple valid ranges for now.

> ReadOnly.msg

This is an example where I don't think a submessage makes any sense or has any benefit. A simple `bool read_only` field in the descriptor serves the same purpose without any other size / complexity overhead.

> Other constraint types could support specific logic for combining.

I would rather not go down the road with combinatoric logic of multiple conditions for now. Keeping it to one constraint at a time for now satisfies what we need it to do and can be extended in the future if we want to.

So how about we start with this:

* append the following fields to `ParameterDescriptor.msg`:
  * `FloatingPointRange[<=1] floating_point_range`
  * `IntegerRange[<=1] integer_range`
  * `bool read_only false`

* create`FloatingPointRange.msg`:
  * `double minimum_value`
  * `double maximum_value`
  * `double step`

* create`IntegerRange.msg`:
  * `int64 minimum_value`
  * `int64 maximum_value`
  * `int64 step 1`

In the future the existing constraints can be extended to become more flexible, new constraints can be added, and multiple constraints of the same type could be allowed with a defined logic.

jacobperron, 2019-04-18:
I like @dirk-thomas's proposal, but it would be nice if there was a way to provide lower and upper bounds independently as @sloretz suggested, e.g. only provide a maximum constraint. Maybe this can be achieved with something like:

```
int64[<=1] minimum_value
int64[<=1] maximum_value
int64 step 1
```

tfoote, 2019-04-18:
@jacobperron How about we make sure that there's a convent way to set MIN_INT64 and MAX_INT64 instead of making the datatype more complex? That will make it clearer to understand and parse with many fewer runtime branch points.

tfoote, 2019-04-18:
Actually reading the XML restrictions logic, maybe we should completely separate the minimum and maximum definitions into separate constraints, then we don't have to worry about the other end if undefined.

dirk-thomas, 2019-04-18:
> How about we make sure that there's a convent way to set MIN_INT64 and MAX_INT64 instead of making the datatype more complex?

There is a semantic difference between providing explicit range boundaries over specifying the numeric limits of the integer type. I would rather not specify the limit if it is not semantically meaningful.

mlanting, 2019-04-18:
What about cases where one parameter is constrained by the current value of another? Maybe not something to try to squeeze in right away, but it's come up for me a number of times when trying to dynamically tune nodes in ROS 1 so I want to put it out there.

dirk-thomas, 2019-04-18:
> What about cases where one parameter is constrained by the current value of another?

A node can already enforce any arbitrary constraint. It is just something which can't be described with a static rule.

wjwwood, 2019-04-18:
That's basically why I suggested a text field that allows the node developer to convey a message that could appear in a GUI, in addition to any of the normal constraints that you can express parametrically, like ranges or choices.

tfoote, 2019-04-18:
+1 for a description of the constraints for debug/logging. An expandable description would be very helpful for people exploring the system. This will also allow a more uniform interaction across different interactions, and have one place to reference for what the logic should be.

dirk-thomas, 2019-04-18:
Can either of you describe how you would extend the proposal from https://github.com/ros2/rcl_interfaces/issues/71#issuecomment-484613435 with a text message? A `string error_message` in the `FloatingPointRange` / `IntegerRange` message?

I am not sure I understand the use case / what values that would be set to (beyond what any generic UI could generate: like "parameter X value Y is smaller than the minimum value of the valid range [MIN, MAX]"). Can you make a concrete example for such an error message?

tfoote, 2019-04-19:
We would add a top level string description of the parameters as a peer of the arrays of constraints.

An example of the text: "" It would have standard min and max -10 and 10 as well as a custom check 

```
string description
MaxConstraint [<=1] max
MinConstraint [<=1] min
```

Example:
```
description: This parameter is required to be in the range -10 to 10 and be higher than parameter foo
max: 10
min: -10
```

The description is what would be printed in the GUI as a tip for the field. And in logging when the parameter fails the runtime validation for being higher than foo.

wjwwood, 2019-04-19:
I was actually thinking of two different strings: a `description` and a `constraint_message` or something like that. But I suppose the description is sufficient. I was just thinking that the description might be too long to put inline in the gui, but short extra message, specifically about the constraints, could be placed near the slider or checkbox or combo box or whatever it has. I was also thinking that you leave the constraint message empty (by convention) if it is just a range. There's no real point in stating that when the slider will indicate that as well. Instead I'd only mention thing in that message that cannot be expressed with the gui easily.

----

> Can either of you describe how you would extend the proposal from #71 (comment) with a text message? A string error_message in the FloatingPointRange / IntegerRange message?

Would it be easier at this point to just open a pull request with the proposed changes which we can comment on inline and have pr's to pr's for proposed alternative representations?

I'll open a pr with what @dirk-thomas proposed, and open my own pr against it with what I'd change. @tfoote can do the same, since I don't quite follow the separate min/max constraint yet.

https://github.com/ros2/rcl_interfaces/pull/72

mlanting, 2019-04-19:
We could call the `description` field `tooltip` instead, to hint to the user how the value will be used.

dirk-thomas, 2019-04-19:
> We could call the `description` field `tooltip` instead,

A `tooltip` is a UI specific concept and in this case overly specific. The description might as well be printed on the console using a CLI tool. Therefore `description` seems to better capture the semantic of the content.

mlanting, 2019-04-19:
Good point, I guess `tooltip` would be more analogous to the `constraint_message` wjwwood had in mind.

wjwwood, 2019-04-20:
Yeah, I think `description` is more appropriate, but I like the intention of `tooltip`. I guess we'll have to hope the tool designers will consider that description may be long and react accordingly, or maybe it will just have to get fixed after iterating with users.

hidmic, 2019-04-30:
Should we close this now that #72 was merged?

closed 2019-04-30 by wjwwood