ros2/rcl_interfaces #75: action_msgs is a magic dependency for new action interfaces

stonier, 2019-04-26:
## Bug report

This is not a bug in the sense that it is broken, rather it is a very awkward thing to debug because it is surprising behaviour with very difficult to grokk error messages when the behaviour isn't followed.

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - crystal
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue & Actual Behaviour

* Create your first action in a typical ros interfaces package
* Compile

```
[ 11%] Generating C introspection for ROS interfaces
/usr/bin/python3 /opt/ros/crystal/lib/rosidl_typesupport_introspection_c/rosidl_typesupport_introspection_c --generator-arguments-file /mnt/mervin/workspaces/crystal/muppet/build/muppet_msgs/rosidl_typesupport_introspection_c__arguments.json
Service interface 'unique_identifier_msgs/UUID' contains an unknown field type: unique_identifier_msgs/UUID action_goal_id
CMakeFiles/muppet_msgs__rosidl_typesupport_introspection_c.dir/build.make:150: recipe for target 'rosidl_typesupport_introspection_c/muppet_msgs/msg/can_frame__rosidl_typesupport_introspection_c.h' failed
make[2]: *** [rosidl_typesupport_introspection_c/muppet_msgs/msg/can_frame__rosidl_typesupport_introspection_c.h] Error 1
make[2]: Leaving directory '/mnt/mervin/workspaces/crystal/muppet/build/muppet_msgs'
make[2]: *** Deleting file 'rosidl_typesupport_introspection_c/muppet_msgs/msg/can_frame__rosidl_typesupport_introspection_c.h'
CMakeFiles/Makefile2:1689: recipe for target 'CMakeFiles/muppet_msgs__rosidl_typesupport_introspection_c.dir/all' failed
make[1]: Leaving directory '/mnt/mervin/workspaces/crystal/muppet/build/muppet_msgs'
Makefile:140: recipe for target 'all' failed
make[1]: *** [CMakeFiles/muppet_msgs__rosidl_typesupport_introspection_c.dir/all] Error 2
make: Leaving directory '/mnt/mervin/workspaces/crystal/muppet/build/muppet_msgs'
make: *** [all] Error 2
```

* Be baffled
* After much digging, guess that `action_msgs` is needed in the  `DEPENDENCIES` of the `rosidl_generate_interfaces` call after seeing the official [example](https://github.com/ros2/example_interfaces/blob/master/CMakeLists.txt#L20) (thanks for that!).

#### Expected behavior

The solution described above is surprising - typically only direct dependencies of variables in messages or services need to be described there. Two ideas:

1. Find a way to bring in `action_msgs` automatically if any actions are to be compiled (if possible)
2. Find a way to provide an error message that recognises the problem and can give a hint of what to do in this situation.

jacobperron, 2019-04-27:
It's possible it was automatic at some point, but regressed? In any case, I think it makes sense for `rosidl_generate_interfaces()` to be aware of the `action_msgs` dependency for action interfaces.

Proposed fix https://github.com/ros2/rosidl/pull/369

jacobperron, 2019-05-30:
ros2/rosidl#369 Now makes it possible to omit `find_package(action_msgs)` and passing it as a dependency to `rosidl_generate_interfaces`. We still require the user to add `<depend>action_msgs</depend>` to their package.xml. Failing to do so will result in an explicit error message telling you to add it.

closed 2019-05-30 by jacobperron