ros2/rcl_interfaces #83: Modern CMake targets

AndreasAZiegler, 2019-07-03:
Please add modern CMake targets support as described in https://pabloariasal.github.io/2018/02/19/its-time-to-do-cmake-right/

ivanpauno, 2019-07-03:
What `CMakelists.txt` file in this repo are you referring to? Could you open a PR with the proposed changes? Thank you!

AndreasAZiegler, 2019-07-05:
I will try to make a PR with my limited knowledge. One question I already have is the following: I'm somehow a bit used to the modern CMake when using add_executable(...), or ad_library(...). As for the ```build_interfaces``` neither is used, I'm kind of stuck how I can apply it here. Any help is appreciated.

ivanpauno, 2019-07-05:
> I will try to make a PR with my limited knowledge. One question I already have is the following: I'm somehow a bit used to the modern CMake when using `add_executable(...)`, or `add_library(...)`. As for the `build_interfaces` neither is used, I'm kind of stuck how I can apply it here. Any help is appreciated.

The `builtin_interfaces` package is only generating some headers and python packages for the `.msg` files in the `msg` folder. For that, it uses `rosidl_generate_interfaces` [macro](https://github.com/ros2/rosidl/blob/9ffe2148b9c249b8539a27745062d6c2c1716f2b/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake#L53-L328). It's not building any executable or library, so it doesn't use `add_executable` or `add_library`.
I think this tutorial can be helpful: https://index.ros.org/doc/ros2/Tutorials/Defining-custom-interfaces-(msg-srv)/.

I'm closing this issue. Feel free to re-open/open another one, if you have any specific suggestion.

closed 2019-07-05 by ivanpauno

RFRIEDM-Trimble, 2022-05-30:
Still missing alias libraries in modern interfaces, blocked by this: 
https://github.com/ros2/rosidl/pull/605