ros2/rcl_interfaces #86: Publishing rcl_interfaces/Log fails

mw-senthil, 2019-10-21:
## Bug report

**Required Info:**

- Operating System:
  - Windows (and all platforms)
- Installation type:
  - binaries 
- Version or commit hash:
  - Dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rcl_interfaces
[rcl_interfaces_problem.zip](https://github.com/ros2/rcl_interfaces/files/3752115/rcl_interfaces_problem.zip)

#### Steps to reproduce issue
Attached is the modified minimal_publisher package to publish rcl_interfaces/Log message.
- unzip in a directory
- colcon build --merge-install
- source install/local_setup.bash
- ros2 run minimal_publisher publisher_lambda 
  - [works]
- ros2 run minimal_publisher publisher_lambda_log
  - [**errors**]
- ros2 run minimal_publisher publisher_lambda_param
  - [works]
```

```

#### Expected behavior
ros2 run minimal_publisher publisher_lambda_log
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 0'
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 1'
....
#### Actual behavior
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 0'
string capacity not greater than size
#### Additional information

Keep seeing an error when program tries to get address of rosidl_typesupport_fastrtps_cpp__get_message_type_support_handle__rcl_interfaces__msg__Log 
Looks like there is some conflict for rcl_interfaces/Log message

jacobperron, 2019-10-22:
Thanks for reporting the issue.

I can reproduce with your example on Ubuntu 18.04:

```sh
$ ros2 run minimal_publisher publisher_lambda_log
[INFO] [minimal_publisher]: Publishing: 'Hello, world! 0'
string capacity not greater than size
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  failed to publish message: cannot publish data, at /home/jacob/ws/latest_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_publish.cpp:52, at /home/jacob/ws/latest_ws/src/ros2/rcl/rcl/src/rcl/publisher.c:290
```

The error persists even if I don't set a message (e.g. the empty string).

Based on the error message, I'm guessing the one of the string fields that is part of [rcl_interfaces/msg/Log](https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/Log.msg) is not being initialized properly. I'll investigate further.

mw-senthil, 2019-11-05:
Any update on this issue? Is there a workaround?

closed 2019-11-05 by dirk-thomas