ros2/rcl_interfaces #87: Failed to load entry point 'launch': No module named 'rcl_interfaces'

galou, 2019-10-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04.
- Installation type:
  - source
- Version or commit hash:
  - 93cedce2dacd25fa4f2969d022ccbe3f2903e3fe
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - the one used in rqt_gui

#### Steps to reproduce issue

```
source ~/ros2_ws/install/local_setup.zsh
ros2 run rqt_gui rqt_gui
```

#### Expected behavior

It should run

#### Actual behavior

```
Failed to load entry point 'test': cannot import name 'remove_ansi_escape_sequences'
Failed to load entry point 'launch': No module named 'rcl_interfaces'
Failed to load entry point 'info': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'send_goal': No module named 'action_msgs'
Failed to load entry point 'show': No module named 'rcl_interfaces'
Failed to load entry point 'play': No module named 'rcl_interfaces'
Failed to load entry point 'record': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'load': No module named 'rcl_interfaces'
Failed to load entry point 'standalone': No module named 'rcl_interfaces'
Failed to load entry point 'types': No module named 'rcl_interfaces'
Failed to load entry point 'unload': No module named 'rcl_interfaces'
Failed to load entry point 'start': No module named 'rcl_interfaces'
Failed to load entry point 'status': No module named 'rcl_interfaces'
Failed to load entry point 'stop': No module named 'rcl_interfaces'
Failed to load entry point 'get': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'nodes': No module named 'rcl_interfaces'
Failed to load entry point 'set': No module named 'rcl_interfaces'
Failed to load entry point 'info': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'delete': No module named 'rcl_interfaces'
Failed to load entry point 'describe': No module named 'rcl_interfaces'
Failed to load entry point 'dump': No module named 'rcl_interfaces'
Failed to load entry point 'get': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'set': No module named 'rcl_interfaces'
Failed to load entry point 'create_key': No module named 'rcl_interfaces'
Failed to load entry point 'create_keystore': No module named 'rcl_interfaces'
Failed to load entry point 'create_permission': No module named 'rcl_interfaces'
Failed to load entry point 'distribute_key': No module named 'rcl_interfaces'
Failed to load entry point 'generate_artifacts': No module named 'rcl_interfaces'
Failed to load entry point 'generate_policy': No module named 'rcl_interfaces'
Failed to load entry point 'list_keys': No module named 'rcl_interfaces'
Failed to load entry point 'call': No module named 'rcl_interfaces'
Failed to load entry point 'find': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'type': No module named 'rcl_interfaces'
Failed to load entry point 'bw': No module named 'rcl_interfaces'
Failed to load entry point 'delay': No module named 'rcl_interfaces'
Failed to load entry point 'echo': No module named 'rcl_interfaces'
Failed to load entry point 'find': No module named 'rcl_interfaces'
Failed to load entry point 'hz': No module named 'rcl_interfaces'
Failed to load entry point 'info': No module named 'rcl_interfaces'
Failed to load entry point 'list': No module named 'rcl_interfaces'
Failed to load entry point 'pub': No module named 'rcl_interfaces'
Failed to load entry point 'type': No module named 'rcl_interfaces'
Traceback (most recent call last):
  File "/home/gael/ros2_ws/install/lib/rqt_gui/rqt_gui", line 6, in <module>
    from rqt_gui.main import Main
  File "/home/gael/ros2_ws/build/rqt_gui/src/rqt_gui/main.py", line 40, in <module>
    import rclpy
  File "/home/gael/ros2_ws/install/lib/python3.6/site-packages/rclpy/__init__.py", line 48, in <module>
    from rclpy.parameter import Parameter
  File "/home/gael/ros2_ws/install/lib/python3.6/site-packages/rclpy/parameter.py", line 18, in <module>
    from rcl_interfaces.msg import Parameter as ParameterMsg
ModuleNotFoundError: No module named 'rcl_interfaces'
```

#### Additional information

ros2.repos from Oct 22 14:00 CET

dirk-thomas, 2019-10-24:
Thanks for the report.

ros2/rclpy#452 adds the missing dependency declaration.

In your case do you have `rcl_interfaces` in your workspace or installed in an underlay? If not, doing so should resolve the problem.

closed 2019-10-24 by dirk-thomas

galou, 2019-10-25:
I have all ros2 packages in a single workspace.

Thanks for the quick fix!