ros2/rcl_interfaces #114: packages.ros.org Foxy package ver. 1.0.0 ships `fibonacci` action from pre-0.7.2

zultron, 2020-11-04:
The `ros-foxy-test-msgs` package currently shipping for Foxy is version `1.0.0-1focal.20201007.204652`.  The package includes headers for the `fibonacci` action, which is very strange, since that action was removed in PR #77, tagged in v. `0.7.2`.

This hasn't directly caused any issues for me (but see ros2/rosidl#544), but it's concerning that the package version doesn't seem to match the source.

## Bug report

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Binary packages
- Version or commit hash:
  - Package `ros-foxy-test-msgs` version `1.0.0-1focal.20201007.204652`
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
$ docker run -it osrf/ros:foxy-desktop bash -xec '
>     apt-get update >&/dev/null
>     apt-get install ros-foxy-test-msgs >&/dev/null
>     dpkg-query -W ros-foxy-test-msgs
>     dpkg-query -L ros-foxy-test-msgs | grep Fibonacci.action
> '
+ apt-get update
+ apt-get install ros-foxy-test-msgs
+ dpkg-query -W ros-foxy-test-msgs
ros-foxy-test-msgs	1.0.0-1focal.20201007.204652
+ dpkg-query -L ros-foxy-test-msgs
+ grep Fibonacci.action
/opt/ros/foxy/share/test_msgs/action/Fibonacci.action
```

#### Expected behavior

The `ros-foxy-test-msgs` v. `1.0.0` package should not contain the `fibonacci` action removed in git tag `0.7.2`.

#### Actual behavior

The `fibonacci` action is included in the package.

#### Additional information

If/when this issue is resolved, at least the [`realtime_tools`][1] and [`nav2_util`][2] packages will break, since they use that action.

[1]: https://github.com/ros-controls/realtime_tools/
[2]: https://github.com/ros-planning/navigation2/

SteveMacenski, 2020-11-04:
Can we get it readded? rclcpp action tests and nav2 utils tests also rely on it. Given `test_msgs` is a tests package, I would think that would be useful to retain

hidmic, 2020-11-04:
@zultron `test_msgs/action/Fibonacci` definition was only moved from the `test_msgs` package to the `test_interface_files` package. It's still there. See [here](https://github.com/ros2/rcl_interfaces/blob/48cb91129051a494f3b4b097dccd6c921bb50552/test_msgs/CMakeLists.txt#L24-L27) and [here](https://github.com/ros2/test_interface_files/tree/foxy).

This doesn't explain https://github.com/ros2/rosidl/issues/544 though.

clalancette, 2020-11-04:
Right.  This is actually expected; we package up the source messages as part of the `test_interface_files`, but we actually build the messages as part of `test_msgs`.  So this is expected here, though I will agree that it is weird setup.

Given that, I'm going to close this one out.  I'll take a look at ros2/rosidl#544 separately.

closed 2020-11-04 by clalancette