ros2/rcl_interfaces #131: Error building rosgraph_msgs

tylerjw, 2021-11-26:
I am trying to follow the [instructions for building ros with tsan](https://github.com/colcon/colcon-sanitizer-reports/blob/master/README.rst) and I'm running into what seems like a really odd build error:

```
--- stderr: rosgraph_msgs                                                                                                                                                                                          
make[2]: execvp: /home/tyler/code/rolling/ws_moveit_source/install-tsan/rosidl_typesupport_cpp/lib/rosidl_typesupport_cpp/rosidl_typesupport_cpp: Permission denied
make[2]: *** [CMakeFiles/rosgraph_msgs__rosidl_typesupport_cpp.dir/build.make:82: rosidl_typesupport_cpp/rosgraph_msgs/msg/clock__type_support.cpp] Error 127
make[1]: *** [CMakeFiles/Makefile2:485: CMakeFiles/rosgraph_msgs__rosidl_typesupport_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
```

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Build from source
  - Version or commit hash:
  - `workspace generated with `rosinstall_generator moveit --rosdistro rolling --deps  | vcs import src`
- DDS implementation:
  - Not specified

#### Steps to reproduce issue
```
colcon build --build-base=build-tsan --install-base=install-tsan \
    --cmake-args -DOSRF_TESTING_TOOLS_CPP_DISABLE_MEMORY_TOOLS=ON \
                 -DINSTALL_EXAMPLES=OFF -DSECURITY=ON --no-warn-unused-cli \
                 -DCMAKE_BUILD_TYPE=Debug \
    --mixin tsan ccache \
    --symlink-install
```

#### Expected behavior
No build error.

#### Actual behavior
If I cd into that directory it looks like it's created a circular symlink:

```
> cd /home/tyler/code/rolling/ws_moveit_source/install-tsan/rosidl_typesupport_c/lib/rosidl_typesupport_c/
> ll
total 4.0K
lrwxrwxrwx 1 tyler tyler 110 Nov 26 13:02 rosidl_typesupport_c -> /home/tyler/code/rolling/ws_moveit_source/src/rosidl_typesupport/rosidl_typesupport_c/bin/rosidl_typesupport_c
```

clalancette, 2021-11-29:
That is pretty odd, I agree.  What happens if you build without `--symlink-install`?

tylerjw, 2021-11-29:
Thank you for the response.  I feel dumb for not trying that.  I just did and it makes it past the rosgraph_msgs package.  I'm happy to assign this to the "symlink-install feature is not ready yet" bucket.  I have had problems with it before and this time I mostly blindly copy-pasted from that guide and somehow didn't notice it (even when the error seemed to be an odd symlink).

clalancette, 2021-12-16:
I'm going to close this out, as this seems to be a problem in `colcon` and we have an issue filed over there.  Feel free to reopen if this is in error.

closed 2021-12-16 by clalancette