ros2/rcl_interfaces #138: Time/Duration comment on nanosec field should limit it to 1e9, not 10e9

Ooorah, 2022-05-06:
## Bug report

**Required Info:**
- Operating System:
  - N/A
- Installation type:
  - N/A
- Version or commit hash:
  - 5a4a3bd4da4cfc35ad377f2966714321b2951c89
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
The comments on the "nanosec" field for both Time and Duration message definitions says:
```
# The nanoseconds component, valid in the range [0, 10e9).
```
The nanoseconds field should be limited to 1e9 (1 billion), non-inclusive, not 10e9 (10 billion). Especially because the data type is uint32, which has a maximum value of about 4.3e9 (4.3 billion).

#### Expected behavior
Comment should say "1e9".

#### Actual behavior
Comment says "10e9".

clalancette, 2022-05-06:
Thanks for the report.

Would you mind opening a pull request to fix it?

Ooorah, 2022-05-06:
Sorry, I've never worked with git before. Just wanted to let you know about the issue.

Ooorah, 2022-05-06:
If I'm understanding the linked pull request correctly, you only fixed Duration.msg. The Time.msg file has the same issue.

tfoote, 2022-05-06:
Thanks I missed that it was both. I fixed it too.

PR #139 (2022-05-12): Fix documented range
Fixes #138
closed 2022-05-12 by clalancette