ros2/rcl_interfaces #142: Feature request: backporting builtin_interfaces to Noetic

ckaran, 2022-07-12:
## Feature request

Due to circumstances beyond my control, I have to make ROS 1 (noetic) and ROS 2 (humble) systems talk with one another.  I've been using the [ros1_bridge](https://github.com/ros2/ros1_bridge), which has worked well so far, but I've hit a stumbling block; [there is no equivalent to `builtin_interfaces` for noetic](https://index.ros.org/p/builtin_interfaces/).  Is it possible to backport it to noetic?

I am reasonably sure that I can make my own backport, but I also know that it will be brittle to do so, and I would prefer to minimize the chances of my creating a conflicting package, which is why I'm asking for the backport.

#### Feature description

As described above, I'd like to see the messages in `builtin_interfaces` backported to noetic.  The only use of this is to make it easier for the [ros1_bridge](https://github.com/ros2/ros1_bridge) to do its job.

#### Implementation considerations

I'm not too sure how to answer this, please forgive my mistakes while trying to.

The easiest/quickest way is for me to simply create my own backport within my project for the `builtin_interfaces` files.  The issue with this is that it's also the fastest way for me to make a mistake such as developing a conflicting definition with the real `builtin_interfaces` files.  

The next most obvious solution is to backport `builtin_interfaces` to noetic.  I don't know how much effort this would be, nor do I know if there would be other conflicts down the road.  I'm relying on the maintainers of this project to decide if the benefit is worth any risks.

tfoote, 2022-07-28:
Can you clarify what your use case is and what isn't working for you?

Time` and `Duration` messages are direct analogies to `[std_msgs/Time](http://docs.ros.org/en/noetic/api/std_msgs/html/msg/Time.html)` and `[std_msgs/Duration](http://docs.ros.org/en/noetic/api/std_msgs/html/msg/Duration.html)` in Noetic. 

The bridge should already know about how to translate those datatypes back and forth as they're used as components in many aggregate messages. 
 The special logic for translating them is embedded here: https://github.com/ros2/ros1_bridge/blob/83390800c5ff205625428f148b514008932cfc61/src/builtin_interfaces_factories.cpp

ckaran, 2022-07-28:
My apologies, I forgot that I'd created this issue, which is why I didn't close it when I'd solved it!

Turns out I was given bad code on my end, once I got it straightened out with the other developer, everything worked.  I'll close this issue now, sorry for the trouble!

closed 2022-07-28 by ckaran