ros2/rcl_interfaces #145: Only representable values should be possible to set as parameters

tylerjw, 2022-10-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - binaries
- Version or commit hash:
  - humble
- DDS implementation:
  - default

#### Steps to reproduce the issue
I reproduced this using the tutorials for moveit2 on Humble.
I found this issue, but I'm unsure if it is related: https://github.com/ros2/rosidl/issues/351

```
Traceback (most recent call last):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rcl_interfaces/msg/_parameter_value.py", line 225, in double_value
    assert value >= -1.7976931348623157e+308 and value <= 1.7976931348623157e+308, \
AssertionError: The 'double_value' field must be a double in [-1.7976931348623157e+308, 1.7976931348623157e+308]

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/humble/bin/ros2", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.18.3', 'console_scripts', 'ros2')())
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2cli/cli.py", line 89, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/command/param.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 118, in main
    pval = self.get_parameter_value(node, absolute_node_name, param_name)
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/verb/dump.py", line 61, in get_parameter_value
    response = call_get_parameters(
  File "/opt/ros/humble/lib/python3.10/site-packages/ros2param/api/__init__.py", line 195, in call_get_parameters
    rclpy.spin_until_future_complete(node, future)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 248, in spin_until_future_complete
    executor.spin_until_future_complete(future, timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 288, in spin_until_future_complete
    self.spin_once_until_future_complete(future, timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 715, in spin_once_until_future_complete
    self.spin_once(timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 712, in spin_once
    raise handler.exception()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 411, in handler
    arg = take_from_wait_list(entity)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 347, in _take_client
    return client.handle.take_response(client.srv_type.Response)
SystemError: <built-in method take_response of PyCapsule object at 0x7f69169317a0> returned a result with an exception set
```

#### Expected behavior

Only representable values should be possible to set as parameters. If it is possible to set a value for a parameter, it should be possible to read that value.

#### Actual behavior

Crash when trying to read parameters

clalancette, 2022-10-07:
> Only representable values should be possible to set as parameters.

Theoretically, this should already be the case.  We have checks in place exactly for this reason.  For instance:

Terminal 1:
```
$ ros2 run turtlesim turtlesim_node
```

Terminal 2:
```
$ ros2 param set /turtlesim background_b 3.4
Setting parameter failed: Wrong parameter type, parameter {background_b} is of type {integer}, setting it to {double} is not allowed.
```

Can you give more details on how to reproduce the error?

tylerjw, 2022-10-07:
I was not able to figure out which parameter this is coming from because my normal way of reading parameters is broken by this. However I was able to write this simple program that builds and runs on Humble and produces the same result:

```c++
#include <cmath>
#include <rclcpp/rclcpp.hpp>

int main(int argc, char * argv[])
{
  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
    "invalid_param",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );
  node->declare_parameter("f", double{NAN});

  rclcpp::spin(node);
  rclcpp::shutdown();
  return 0;
}
```

clalancette, 2022-10-12:
OK, I just confirmed that yes, this is the same problem as https://github.com/ros2/rosidl/issues/351 .  I also confirmed that this is not a problem on Rolling anymore.  So I'm going to close this out as a dup.

closed 2022-10-12 by clalancette