ros2/rcl_interfaces #167: action_msgs is missing dependency on service_msgs

christianrauch, 2024-09-14:
## Bug report

I am trying to build a workspace from scratch:
```yaml
repositories:
  ament/ament_cmake:
    type: git
    url: https://github.com/ament/ament_cmake.git
    version: 2.5.2
  ament/ament_index:
    type: git
    url: https://github.com/ament/ament_index.git
    version: 1.8.1
  ament/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: 0.17.1
  ament/ament_package:
    type: git
    url: https://github.com/ament/ament_package.git
    version: 0.16.3

  ros2/ament_cmake_ros:
    type: git
    url: https://github.com/ros2/ament_cmake_ros.git
    version: 0.12.0
  ros2/rcl_interfaces:
    type: git
    url: https://github.com/ros2/rcl_interfaces.git
    version: 2.0.2
  ros2/rcutils:
    type: git
    url: https://github.com/ros2/rcutils.git
    version: 6.7.1
  ros2/rosidl:
    type: git
    url: https://github.com/ros2/rosidl.git
    version: 4.6.3
  ros2/rosidl_core:
    type: git
    url: https://github.com/ros2/rosidl_core.git
    version: 0.2.0
  ros2/rosidl_defaults:
    type: git
    url: https://github.com/ros2/rosidl_defaults.git
    version: 1.6.0
  ros2/unique_identifier_msgs:
    type: git
    url: https://github.com/ros2/unique_identifier_msgs.git
    version: 2.5.0
```

Running `colcon build` will fail with:
```
--- stderr: action_msgs                                                                                                                                         
CMake Error at [...]/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:178 (message):
  Unable to generate service interface for 'srv/CancelGoal.srv'.  In order to
  generate service interfaces you must add a depend tag for 'service_msgs' in
  your package.xml.
Call Stack (most recent call first):
  CMakeLists.txt:30 (rosidl_generate_interfaces)
```

But `<depend>service_msgs</depend>` is already listed in the `package.xml`.

**Required Info:**

- Operating System:
  - Ubuntu
- Installation type:
  - source
- Version or commit hash:
  - 2.0.2 / jazzy

#### Steps to reproduce issue
See above.

#### Expected behavior

#### Actual behavior

fujitatomoya, 2024-09-16:
@christianrauch can you provide the full repo file or commit hash from https://github.com/ros2/ros2/tree/jazzy?

i tries with your repo file, but failed to build with the following error. i believe many packages are missing to complete the build.

```bash
root@tomoyafujita:~/ros2_ws/work# colcon build --symlink-install --packages-up-to action_msgs
...
--- stderr: rcutils
CMake Error at CMakeLists.txt:162 (find_package):
  By not providing "Findperformance_test_fixture.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "performance_test_fixture", but CMake did not find one.

  Could not find a package configuration file provided by
  "performance_test_fixture" with any of the following names:

    performance_test_fixtureConfig.cmake
    performance_test_fixture-config.cmake

  Add the installation prefix of "performance_test_fixture" to
  CMAKE_PREFIX_PATH or set "performance_test_fixture_DIR" to a directory
  containing one of the above files.  If "performance_test_fixture" provides
  a separate development package or SDK, be sure it has been installed.

---
Failed   <<< rcutils [0.35s, exited with code 1]
Aborted  <<< rosidl_adapter [0.22s]

Summary: 45 packages finished [5.31s]
  1 package failed: rcutils
  1 package aborted: rosidl_adapter
  15 packages had stderr output: ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_package ament_pep257 ament_pycodestyle ament_uncrustify ament_xmllint domain_coordinator rcutils rosidl_cli
  16 packages not processed
```

Mechazo11, 2024-10-09:
@christianrauch have you fixed this error, I too got this error for the ```jazzy``` branch.

```
--- stderr: rcl_interfaces                                                    
CMake Error at /home/tigerwife/mppi_rose25_ws/install/rosidl_cmake/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:178 (message):
  Unable to generate service interface for 'srv/DescribeParameters.srv'.  In
  order to generate service interfaces you must add a depend tag for
  'service_msgs' in your package.xml.
Call Stack (most recent call first):
  CMakeLists.txt:18 (rosidl_generate_interfaces)
```

christianrauch, 2024-10-14:
> @christianrauch have you fixed this error, I too got this error for the `jazzy` branch.

No. I guess this issue will stay open until a PR with a fix has been merged.