ros2/rcl_interfaces #175: Duration.msg and Time.msg nanoseconds description

jimmy-mcelwain, 2025-04-04:
In both the duration message and time message definitions the `seconds` description is (roughly) as follows:

`Seconds component, range is valid over any possible int32 value.`

And the `nanoseconds` description is:

`Nanoseconds component in the range of [0, 1e9).`

So, the seconds description seems pretty explicit that it can be negative. My assumption is that this generally means that:

1. A time message can represent a time before the epoch
2. A duration message can represent a negative duration

Those make sense to me. 

But the nanoseconds description is weird to me. It says it is the "nanoseconds component". That makes sense for time and duration values with positive `seconds` values, but for a time or duration with a negative `seconds`, how does that work?

Specifically, how would a number between 0 and -1 be represented? Given that `nanoseconds` is of type `uint32`, it cannot be negative. But `seconds` is of type `int32`, and in C cannot represent negative zero. 

For example, take `-0.1 seconds`. How would it be represented? 

`seconds: 0, nanoseconds: -100000000` -- doesn't work because nanoseconds must be positive
`seconds: -0, nanoseconds: 100000000` -- doesn't work because negative zero isn't valid in C/isn't in the domain of int32. 
`seconds: -1, nanoseconds: 900000000` -- this would work, but goes against the description `Nanoseconds component in the range of [0, 1e9).`. It's also not very intuitive, but that's alright I guess if it works.

What is the expected way to represent this in C?

christophebedard, 2025-04-05:
I think you figured it out, but see also:

1. Duration: https://github.com/ros2/rclcpp/blob/892cae99159597b85d3e569953ed67652a0c9eed/rclcpp/src/rclcpp/duration.cpp#L52-L54
2. Time: https://github.com/ros2/rclcpp/blob/892cae99159597b85d3e569953ed67652a0c9eed/rclcpp/src/rclcpp/time.cpp#L83-L84

Updating the comments in the messages to make this clearer or provide an example would probably be helpful. Maybe just:

```diff
-# The nanoseconds component, valid in the range [0, 1e9).
+# The nanoseconds component to add to the second component, valid in the range [0, 1e9).
```

jimmy-mcelwain, 2025-04-07:
Thank you. I think changing the comment to something like that or adding an example would be helpful.

christophebedard, 2025-04-07:
Would you be willing to submit a PR?

> `seconds: -1, nanoseconds: 900000000` -- this would work, but goes against the description `Nanoseconds component in the range of [0, 1e9).`. It's also not very intuitive, but that's alright I guess if it works.

It works and doesn't go against the comment, because 1e9 nanoseconds is 1 second.

PR #176 (2025-05-09): Add info to duration message and time message comments
Closes #175

Makes comments for duration messages and time messages more clear, and adds examples for the message types.
PR #177 (2025-05-09): Add info to duration message and time message comments (backport #176)
Closes #175

Makes comments for duration messages and time messages more clear, and adds examples for the message types. <hr>This is an automatic backport of pull request #176 done by [Mergify](https://mergify.com).
PR #178 (2025-05-09): Add info to duration message and time message comments (backport #176)
Closes #175

Makes comments for duration messages and time messages more clear, and adds examples for the message types. <hr>This is an automatic backport of pull request #176 done by [Mergify](https://mergify.com).
PR #179 (2025-05-09): Add info to duration message and time message comments (backport #176)
Closes #175

Makes comments for duration messages and time messages more clear, and adds examples for the message types. <hr>This is an automatic backport of pull request #176 done by [Mergify](https://mergify.com).
closed 2025-05-09 by ahcorde